aws/ota-for-aws-iot-embedded-sdk #201: Build error in C ++ projects

ser90368141, 2021-04-01:
I am building a library for ESP32S2. The main project for ESP32S2 is written in C ++. 
While compiling the library, I get the error:
**ota_os_interface.h: 272: 22: error: expected unqualified-id before 'delete'**
As I understand it, this is due to the fact that in C ++, delete is a keyword. Therefore, it cannot be used as a variable name.

https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/a1794596f0ba94e5f422cb064e7c60e8218f1485/source/include/ota_os_interface.h#L272

After changing the name, the assembly of the library was successful and the library does not cause problems.

b1ackviking, 2021-04-01:
I'm facing the same issue. Unfortunately, the library headers can't compile in a C++ project for such a stupid reason.

pvyawaha, 2021-04-01:
Thank you for reporting this issue, we are working on fixing this.

archigup, 2021-04-05:
Closing this issue as the fix is merged. Thanks!

closed 2021-04-05 by archigup