aws/ota-for-aws-iot-embedded-sdk #204: OS interface: Inconsistency in the receive timeout parameter

boraozgen, 2021-04-21:
There seems to be an inconsistency in the `OtaReceiveEvent_t` documentation and its usage. Although the signature defines a timeout parameter, this seems to be called with 0 timeout all the time by `OTA_EventProcessingTask`, effectively spamming the thread and not letting it block. This was implemented with `mq_receive()` without using the timeout parameter in the demo, which blocks indefinitely. Please clarify the interface with appropriate documentation.

pvyawaha, 2021-04-22:
Hello , thank you for pointing this out. The OTA OS interface is designed with the timeout parameter but it is not used when calling the receive event function from the OTA library. The OTA_EventProcessingTask drives the main OTA state machine and if timeout is used the loop will call receive event again after timeout expires. We will work on the documentation to make it more clear.

ActoryOu, 2022-07-07:
Hi @boraozgen ,
We've updated the [Posix_OtaReceiveEvent/Posix_OtaSendEvent](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/main/source/portable/os/ota_os_posix.c#L134-L252) and [OtaSendEvent_FreeRTOS/OtaReceiveEvent_FreeRTOS](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/main/source/portable/os/ota_os_freertos.c#L93-L154) to match the description in documentation.

I'm going to close this issue.
Thanks again for your contribution!

closed 2022-07-07 by ActoryOu