aws/ota-for-aws-iot-embedded-sdk #206: Feature request: Store download state in persistent memory

boraozgen, 2021-04-22:
Downloads usually take a long time and data costs are high with low-bandwidth networks, such as NB-IoT. It would be nice to have a possibility to save the current download state to be able to continue later without needing to repeat the already downloaded blocks, e.g. in case of a reset/power loss. Is such a feature planned?

pvyawaha, 2021-04-24:
Hello , The library supports resuming a download but it does not store the context in NVM memory . We are reviewing the feature requests but at this time I do not have a release data for this feature. When file blocks are received they are stored in a bitmap, resetting a bit corresponding to the file block. This needs to be stored in the NVM and restored when it boots up.

n9wxu, 2023-12-20:
No further development will be performed on this library. See the readme.

closed 2023-12-20 by n9wxu