aws/ota-for-aws-iot-embedded-sdk #218: OTA_SignalEvent has undefiend behaviour

fhars, 2021-05-25:
This line:
https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/0d6f669d4348a7130811c4d7c260218aea6534c2/source/ota.c#L2933
may dereference a null pointer, especially after an `OTA_Shutdown(ticks, false)`. (Note that checking that pOtaInterface is valid before the call is not sufficient here, as the `shutdownHandler` might memset `otaAgent` to zero between the check and the call). Another possible race condition is in the called `send` method, as (if the FreeRTOS case) the queue itself might be deinitialized by the time execution reaches the `configAssert( pxQueue );` in `xQueueGenericSend`, although this error path is currently masked by #217.

fhars, 2021-05-25:
The same thing in `OTA_Suspend`:
```
    /* Check if OTA Agent is running. */
    if( otaAgent.state != OtaAgentStateStopped )
    {
        /* Stop the request timer. */
        ( void ) otaAgent.pOtaInterface->os.timer.stopTimer( OtaRequestTimer );
```
and in OTA_ActivateNewImage (this time with the insufficient check):
```
    if( ( otaAgent.pOtaInterface != NULL ) && ( otaAgent.pOtaInterface->pal.activate != NULL ) )
```

divekarshubham, 2021-09-01:
Thank you for bringing these issues to our attention. We will work on adding further checks for these functions.

lightblu, 2022-01-12:
+1

Want to add that I believe this doesn't even need such a hard/tight preemptive race like theorized above, but could even happen with e.g. cooperative FreeRTOS timer race, because a timer may just expire when `OTA_Shutdown` is called, and the timer callback then stills gets called afterwards.

That was my theory for some time, until I realized [OTA_Shutdown does not clear started timers at all](https://github.com/aws/ota-for-aws-iot-embedded-sdk/issues/412).

_I feel bad for complaining on an open source project like this, but as this is the advertised library by AWS I have to leave the snarky comment that I'm quite disappointed that this very ugly system crashing issue has not been touched for over half a year now, while at the same time endless CBMC proofs are added for other places ( reading [CBMC verifies memory safety (which includes array bounds checks and checks for the safe use of pointers)](https://www.cprover.org/cbmc/) after having to figure out that one feels funny..)_

archigup, 2022-01-18:
Hi, apologies for the delay. OTA_Shutdown has now been updated to clear started timers here: https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/413.
We are investigating additional fixes needed to resolve this issue.

lightblu, 2022-01-21:
Thanks. 
Also want to point out, there is even another simple path how to get this undefined behaviour, which is via the packet ingestion. So imagine you have called OTA_Shutdown, but your Mqtt module is still receiving a late OTA message it then passes on to the OTA library via OTA_SignalEvent.. 

(One could sure say you shouldn't do pass packets to OTA after you have called the shutdown method, but better would be to also guard this path. It at least needs to be documented, and there I also have to agree with fhars' https://github.com/aws/ota-for-aws-iot-embedded-sdk/issues/224 - it is not only not well abstracted but also not at all documented. "So one has to rifle through the source of the library" also matches my memory of getting that part right.

ActoryOu, 2022-06-23:
PR #440 provides the solution for this. Once the OTA_Init called, the function pointer is set and never reset even it's shutdowned.
So there shouldn't be any NULL function pointer accessed with this patch.

I'm going to close this issue. Any feedback is welcome.

Thanks.