aws/ota-for-aws-iot-embedded-sdk #220: AppCallback API usage in all examples is not const correct

fhars, 2021-05-25:
The second parameter of `OtaAppCallback_t` is defined as `const void * pData` , but then the only documented uses of that parameter are to use it for memory management purposes and cast the constness away:

https://github.com/FreeRTOS/FreeRTOS/blob/c134a581153c1b137af2af2e93564c495977872f/FreeRTOS-Plus/Demo/AWS/Ota_Windows_Simulator/Ota_Over_Mqtt_Demo/DemoTasks/OtaOverMqttDemoExample.c#L836

https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/5eec3f03f420b904e1c041e5bda3eeebb6f5744b/demos/ota/ota_demo_core_mqtt/ota_demo_core_mqtt.c#L664

Also, if some messages happen to be in the queue the moment the agent decides to go into shutdown, the callback is never called for these packages and the buffers leak.

pvyawaha, 2021-09-01:
Hello , 

We are working on adding more documentation and examples for the `OtaAppCallback_t `. The buffers are in the demos and not the library so it can be cleared in the demo/application and we will add that in the ota demos in the repository.

ActoryOu, 2022-06-23:
Hi,
Thanks again for contribution.
We removed the const property in the OtaAppCallback_t function prototype with PR #443.
And we also provided more description for this function usage.

I'm going to close this issue. Any feedback is welcome.

Thanks.

closed 2022-06-23 by ActoryOu