aws/ota-for-aws-iot-embedded-sdk #223: OTA agent should not update non-OTA jobs as failed

boraozgen, 2021-05-26:
The OTA agent currently updates the non-OTA jobs as failed:

https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/1938dd89dc198fec37f55a4e53780561c1cfa2c8/source/ota.c#L2245-L2259

This interferes with the usage of non-OTA jobs and IMO not desired. Is there a reason for this logic?

dachalco, 2021-05-26:
Hi @boraozgen 

Thanks for bringing this to our attention. I've notified an appropriate team mate. There should be another response soon.

pvyawaha, 2021-09-01:
Hello @boraozgen, 

We are working on a PR to make this configurable/optional as OTA library cane be used for custom jobs as well.

ActoryOu, 2022-06-13:
Hi @boraozgen,
Thanks for your asking.
I tried to test with custom job and I can pass the part by setting "pJobDoc->parseErr = OtaJobParseErrNone;" when I received OtaJobEventParseCustomJob in application callback function (set by OTA_Init). Please try to do it if you valid the json format.

Example.
```
static void prvOtaAppCallback( OtaJobEvent_t xEvent,
                               const void * pData )
{
    OtaJobDocument_t * pJobDoc = NULL;
    switch( xEvent )
    {
        case OtaJobEventParseCustomJob:
            LogInfo( ( "Received OtaJobEventParseCustomJob from OTA Agent." ) );
            pJobDoc = (OtaJobDocument_t*) pData;
            pJobDoc->parseErr = OtaJobParseErrNone;
            break;
    }
    ...
}
```

Log by ESP32.
> 79 2087 [OTA Agent Task] [ota.c:1649] [INFO] [OTA] Extracted parameter: [key: value]=[execution.jobId: customTest]
> 80 2088 [OTA Agent Task] [ota_demo_core_mqtt.c:940] [INFO] [OTA] Received OtaJobEventParseCustomJob from OTA Agent.
> 83 2137 [MQTT Agent Task] [core_mqtt_agent_command_functions.c:76] [INFO] [MQTT] Publishing message to $aws/things/MqttTest/jobs/customTest/update.
> 84 2138 [OTA Agent Task] [ota_demo_core_mqtt.c:1575] [INFO] [OTA] Sent PUBLISH packet to broker $aws/things/MqttTest/jobs/customTest/update to broker.
> 85 2139 [OTA Agent Task] [ota.c:1999] [INFO] [OTA] Job document parsed from external callback

Since I can pass the custom job without any code change on OTA, I'll close this issue first.
Please re-open it if you need more help from us.

Thanks.

closed 2022-06-13 by ActoryOu

boraozgen, 2022-06-13:
Well that is a hacky way to do it. 

I solved this by filtering the jobs responses before they are dispatched to the OTA client. If the `jobDocument` contains `afr_ota`, they get dispatched to the OTA client, otherwise to the respective job handler.

ActoryOu, 2022-06-13:
That is also a good method to do it, but you'll need to parse the job twice (one before dispatching, the other in OTA).
And I think the method I mentioned is the common method for custom job. (refer to [unit-test code](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/main/test/unit-test/ota_utest.c#L717-L721))

There is no more description in official document, and I'll update it.

Thanks for your contribution.