aws/ota-for-aws-iot-embedded-sdk #238: Change the default signature size to support RSA3072 or RSA4096

NickMcKendree, 2021-08-02:
As of May 2021, Digicert has stopped issuing 2048 bit Code Signing Certificates and has moved to 3072 bit Certificates. Their notification page indicates this is an industry wide standards change so the AWS OTA library should follow the change. 

The default setting of the OTA library is 256 bytes, which only supports RSA-2048. I pulled the latest main to verify that this is still the default setting. Unless it is a breaking issue somewhere else, should kOTA_MaxSignatureSize be defined for RSA4096 to remain flexible for future changes (who knows how long RSA3072 will remain the standard). 

The change is simple...
In ota_private.h, 
`#define kOTA_MaxSignatureSize           512/* Max bytes supported for a file signature (4096 bit RSA is 512 bytes). */`
or
`#define kOTA_MaxSignatureSize           384/* Max bytes supported for a file signature (3072 bit RSA is 384 bytes). */`

https://knowledge.digicert.com/alerts/code-signing-new-minimum-rsa-keysize.html

pvyawaha, 2021-08-04:
Hello NickMcKendree,

Thank you for bringing this to notice, we are working on a change for this.

ActoryOu, 2022-06-15:
The default signature size is changed to 384 to support RSA-3072 after #432 merged.
I'm going to close this issue. Thanks for your reminding.

closed 2022-06-15 by ActoryOu