aws/ota-for-aws-iot-embedded-sdk #323: Meet invalid path ':' on git-checkout on Windows

ccli8, 2021-10-01:
Hi,

I meet invalid path ':' when I run git to clone and checkout this repository on Windows (git version 2.33.0.windows.2):

```
$ git clone https://github.com/aws/ota-for-aws-iot-embedded-sdk
Cloning into 'ota-for-aws-iot-embedded-sdk'...
remote: Enumerating objects: 2489, done.
remote: Counting objects: 100% (878/878), done.
remote: Compressing objects: 100% (403/403), done.
Reremote: Total 2489 (delta 672), reused 571 (delta 474), pack-reused 1611
Receiving objects: 100% (2489/2489), 1.62 MiB | 3.03 MiB/s, done.
Resolving deltas: 100% (1653/1653), done.
error: invalid path ':'
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

It seems that there are file names with ':' characters added into `main` branch, which are invalid for WIndows NTFS. Would you rename such files so that this repository gets available for Windows users?

Bujain, 2021-10-05:
Hello ccli8, 

Thank you for bringing out this issue. We are working on a fix for it.

gshvang, 2021-10-06:
Hi @ccli8 ,

We have merged the fix for the issue. Please reach out to us if you have any further questions.

Thanks

closed 2021-10-06 by gshvang