aws/ota-for-aws-iot-embedded-sdk #411: Is it possible to use OTA library to download file via custom job?

itwondersteam, 2022-01-05:
It's understood that the library is built for firmware update, however is it possible to use it to perform file download via coreHTTP? As otherwise we would have to implement the HTTP download mechanism again just for that, though most of the coreHTTP parts are already implemented in the OTA library... so we are hoping to make use of it if possible.

pvyawaha, 2022-01-05:
Hello @itwondersteam ,

You can use the OTA library for other types of file download which are not firmware files. The default demo and configuration assigns file type ID 0 for firmware files and you can use other file type IDs for downloading other files. You will have to update the OTA PAL for your platform to create and write the received file blocks based on the file type ID received in the job document and available in the OTA context passed to all OTA PAL functions.

itwondersteam, 2022-01-06:
Thanks, yap, it does get the file to be downloaded when we set the file type to 2. However, if creating a new job with same file again, this will trigger `Rejecting new image and rebooting:The platform is in the self-test state while the job is not.`, there's a file PlatformImageState.txt get created and deleting it allow the file to be downloaded again.

Is there anything we need to set so that it does not trigger the platform to be in the self test state?

pvyawaha, 2022-01-06:
Hello @itwondersteam .

Can you please share which OTA PAL you are using?

itwondersteam, 2022-01-07:
Currently it is developed in Linux platform, so using POSIX PAL from [ota_pal_posix.c](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/platform/posix/ota_pal/source/ota_pal_posix.c)

```
    pOtaInterfaces->pal.getPlatformImageState = otaPal_GetPlatformImageState;
    pOtaInterfaces->pal.setPlatformImageState = otaPal_SetPlatformImageState;
    pOtaInterfaces->pal.writeBlock = otaPal_WriteBlock;
    pOtaInterfaces->pal.activate = otaPal_ActivateNewImage;
    pOtaInterfaces->pal.closeFile = otaPal_CloseFile;
    pOtaInterfaces->pal.reset = otaPal_ResetDevice;
    pOtaInterfaces->pal.abort = otaPal_Abort;
    pOtaInterfaces->pal.createFile = otaPal_CreateFileForRx
```

itwondersteam, 2022-01-12:
@pvyawaha is it possible to configure this clean? currently we listen to `OtaJobEventUpdateComplete` then check the filetype and delete the PlatformImageState.txt. But this feels not so proper.

pvyawaha, 2022-01-14:
Hello @itwondersteam,

The ota_pal_posix.c is reference implementation and not part of the library. You can update it as per your application requirement. On platforms where you are receiving other than firmware file ( which requires reboot and self-test ) you can check the file type id and handle the updates to PlatformImageState.txt. For example you can directly set the image state to accepted/valid instead of testing after the crypto signature verification is successful.

itwondersteam, 2022-01-17:
Thanks @pvyawaha  for the pointer, that solved our issue!

closed 2022-01-17 by itwondersteam