aws/ota-for-aws-iot-embedded-sdk #412: OTA_Shutdown does not clear timers (then leading to null pointer dereference)

lightblu, 2022-01-12:
In contrast to the docs for `OTA_Shutdown` stating "and clear all resources." that does not seem to care for any running timers.

This has been observed and reproduced when the MQTT operation for requesting a new job fails after OTA agent startup (e.g. due to disconnection or a shutdown of the connection meanwhile) which makes the OTA lib start the `OtaRequestTimer`. 
And if then `OTA_Shutdown` is called, that leaves the timer running beyond OTA agent having shut down.

This is especially bad as `OTA_Shutdown` clears the whole `otaAgent`struct, including the `pOtaInterface`pointer.
So when the timer fires, it goes via `otaTimerCallback` into `OTA_SignalEvent` which then does a `otaAgent.pOtaInterface->os.event.send( NULL, pEventMsg, 0 );`

Besides this bad crash (kind of sad and funny: the os.event.send function is the second function in the whole struct, thus indexes to a 0x4 from `pOtaInterface`, so if `pOtaInterface` is `NULL`, this actually calls the after-reset-entry-point on our Kinetis ARM M4s, so this didn't even look like the usual rare HardFault but disguised as a very strange rare unexpected reset :( ), this will memory leak created timers.

Note, `OTA_Suspend` does a `otaAgent.pOtaInterface->os.timer.stop( OtaRequestTimer );`, but not `OTA_Shutdown`. However  glancing over the code, it seems this scenario would also be true for the second `OtaSelfTestTimer` which is not even cleared by `OTA_Suspend`, if this is revisited, the whole timer logic needs to be checked.

(Further, I'd appreciate if `ota_os_freertos.c` would make use of static timers not only to avoid dynamic memory leaks on another level, but also because this is one of the last things that would allow the mqtt version to run without heap).

_(This has been observed and confirmed on a 3.1.0 from July - I see though nothing in latest develop til now that touched this)._

archigup, 2022-01-12:
Hi, we are looking into the issue; I have made a PR to add clearing timers.

Would you be able to make a separate issue for using static timers?

archigup, 2022-01-18:
Hi, https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/413 adds clearing the timers to OTA_Shutdown. Apologies for the delay.

closed 2022-01-18 by archigup

lightblu, 2022-01-21:
Thank you!

> Would you be able to make a separate issue for using static timers?

Created https://github.com/aws/ota-for-aws-iot-embedded-sdk/issues/415