aws/ota-for-aws-iot-embedded-sdk #415: Remove dynamic memory allocation from ota lib

lightblu, 2022-01-21:
This is just a not high priority feature enhancement request;

It would be very nice in environments where dynamic memory allocation is forbidden (coming from https://github.com/aws/ota-for-aws-iot-embedded-sdk/issues/412#issue-1100132314 )

There seem to be not much places that would need to change, one is in `ota_os_freertos.c` for the timer creation, and then there are only thre os.mem.malloc's in `ota.c` hat also look to have only limtied used and seem like they could be done static as well without wasting too much RAM - this could also be just a transparent define/compile option for the ota_os_freertos port.

dachalco, 2022-01-21:
Hi @lightblu 

This seems like a great suggestion. I'll inspect the code to verify feasibility and see what changes are needed.

pvyawaha, 2022-01-25:
Hello @lightblu ,

The OTA library can be used without any dynamic allocation and os.mem.alloc is provided optional. You can set the buffers while initializing the OTA agent. 

The ota_os_freertos.c is reference implementation for freeRTOS, not part of OTA library and you can modify it as per your requirements for example timer creation.

pvyawaha, 2022-03-11:
Closing this topic now, please reopen for same topic or create new one for others.