aws/ota-for-aws-iot-embedded-sdk #418: jobDoc.pJobId is memset to zero before dispatching to OtaAppCallback

kenng, 2022-02-17:
Probably can memset the jobId after calling OtaAppCallback? As user application unable to retrieve the jobId.

```c
static OtaErr_t processDataHandler( const OtaEventData_t * pEventData )
{
    // ...

    /* Set the job id and length from OTA context. */
    jobDoc.pJobId = otaAgent.pActiveJobName;
    jobDoc.jobIdLength = strlen( ( const char * ) otaAgent.pActiveJobName ) + 1U;
    jobDoc.fileTypeId = otaAgent.fileContext.fileType;

    if( result == IngestResultFileComplete )
    {
        /* Check if this is firmware update. */
        if( otaAgent.fileContext.fileType == configOTA_FIRMWARE_UPDATE_FILE_TYPE_ID )
        {
            jobDoc.status = JobStatusInProgress;
            jobDoc.reason = JobReasonSigCheckPassed;

            otaJobEvent = OtaJobEventActivate;
        }
        else
        {
            jobDoc.status = JobStatusSucceeded;
            jobDoc.reason = JobReasonAccepted;
            jobDoc.subReason = ( int32_t ) otaAgent.fileContext.fileType;

            otaJobEvent = OtaJobEventUpdateComplete;
        }

        /* File receive is complete and authenticated. Update the job status. */
        err = otaControlInterface.updateJobStatus( &otaAgent, jobDoc.status, jobDoc.reason, jobDoc.subReason );

        // otaAgent.pActiveJobName set to 0 before dispatch to OtaAppCallback
        dataHandlerCleanup();

        if( otaAgent.statistics.otaPacketsProcessed < UINT32_MAX )
        {
            /* Last file block processed, increment the statistics. */
            otaAgent.statistics.otaPacketsProcessed++;
        }

        /* Let main application know that update is complete */
        otaAgent.OtaAppCallback( otaJobEvent, &jobDoc );
```

n9wxu, 2022-02-17:
Thank you for reporting this issue.  It looks like some adjustments can be made here.  We are taking a look at it.

archigup, 2022-02-18:
Pr to move the memset was merged here: https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/419

kenng, 2022-02-21:
done a quick test, and able to get the jobID back. Thanks for the prompt response!

closed 2022-02-21 by kenng