aws/ota-for-aws-iot-embedded-sdk #420: FreeRTOS timer interface: deleted timers started after shutdown lead to null pointer dereference

TJKrawczynski2, 2022-03-17:
As in issue https://github.com/aws/ota-for-aws-iot-embedded-sdk/issues/412, timers that are started before the OTA agent runs the `shutdownHandler` (which zeros `pOtaInterface`) will enter `OTA_SignalEvent` from `otaTimerCallback` and then attempt to dereference `otaAgent.pOtaInterface->os.event.send( NULL, pEventMsg, 0 );`

This issue was marked as resolved following the merging of https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/413, however the issue still persists in the special case where timers are started _after_ OTA_Shutdown is called. 

I was able to reproduce this like so:
- update to patch: https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/413
- request and receive a bunch of file blocks
- before all blocks are digested, call OTA_Shutdown
  - note that all timers are stopped and "deleted" here
- on processing the last requested block, `processDataHandler` calls `otaAgent.pOtaInterface->os.timer.start`
  - os.timer.start calls `OtaStartTimer_FreeRTOS` which will re-create the timer that was deleted in `OTA_Shutdown`
- the `OtaAgentEventShutdown` event gets processed and the ota agent is zeroed
- timer callback expires and dereferences a null pointer.

This may be a problem with the way the FreeRTOS os interface is set up, but also hints that there is some ambiguity with how OTA timer interface APIs should be implemented.

ActoryOu, 2022-06-15:
Hi,
This issue can be solved by PR #440 because we don't clear the function pointer even after OTA_Shutdown.
And OTA drops all the events in the next OTA_Init to make sure there is no remaining events inside the event queue.

Thanks for your contribution.

TJKrawczynski2, 2022-06-23:
Thanks @ActoryOu, has there been any attempt to reproduce the issue on the new patch (#440)?