aws/ota-for-aws-iot-embedded-sdk #434: ESP32 internet disconnect

git4gaurav, 2022-05-22:
Hi, the framework given by AWS for ESP32-DevkitC based OTA MQTT is crashing when internet disconnects. During crash, 6 out of 5 retry for TLS attempts is seen in serial log. Please suggest how to overcome this. If any other info required please let me know.

ravibhagavandas, 2022-05-24:
Hi,
Could you provide more information about the issue ? Is the crash seen in the application or the OTA library?
Can you use a debugger to know the exact location of the crash or a stack trace? More information of ESP32 debugging can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html
Could you also share the serial terminal logs?

git4gaurav, 2022-05-31:
Hi Ravi, Thanks for reply and sorry for delay in response. Please check attached file having serial log. OTA MQTT from AWS IoT framework working ok, but when hotspot used from mobile for cloud connection is closed, then crash is seen on device after 6 of 5 attempts of re-try. Please check log, text highlighted where reebot happens.
[awsFrameworkOtaCrash.txt](https://github.com/aws/ota-for-aws-iot-embedded-sdk/files/8804393/awsFrameworkOtaCrash.txt)

ravibhagavandas, 2022-05-31:
I am not sure the exact demo code (which version of the demo) are you referring to. But looking at the MQTT Agent task function in the latest demo [here](https://github.com/aws/amazon-freertos/blob/main/demos/ota/ota_demo_core_mqtt/ota_demo_core_mqtt.c#L1693) the demo function asserts and does not continue after all retries are exhausted. The assert is causing the reboot. You can replace the assert with your own application level logic to perform when it cant connect to a down network.
Also you can increase the number of retries, if you think your application should do more number of connection attempts before giving up.

https://github.com/aws/amazon-freertos/blob/main/demos/ota/ota_demo_core_mqtt/ota_demo_core_mqtt.c#L263

aggarg, 2022-06-03:
I am closing this issue. Feel free to reopen or create a new one if you still face any issue.

closed 2022-06-03 by aggarg