aws/ota-for-aws-iot-embedded-sdk #437: OTA in multiple device

git4gaurav, 2022-05-23:
OTA for a single device is working. Please suggest how can I program multiple devices with one OTA job. Single binary file should be programmed in selected number of devices through OTA. Is this possible using Thing group? Please suggest.

Thanks

n9wxu, 2022-05-23:
When you create the OTA job in the console you are given the choice to select your things, or select your thing group.  You can select multiple items in the list including a mix of things and thing groups.
[image]

ravibhagavandas, 2022-05-29:
Hope this clarifies your query. Please feel to open an issue if you have further questions.

closed 2022-05-29 by ravibhagavandas