aws/ota-for-aws-iot-embedded-sdk #442: random Block ID

CodeOn-ArK, 2022-06-12:
Hi there!
So when I was watching the logs I saw that the AWS sends blocks in random manner. The block IDs are not serial in fashion. It even misses out several numbers in the middle. So are the payloads sent randomly as the block IDs of the payloads are random, or there's some algorithm behind it. 
Regards

ravibhagavandas, 2022-06-13:
Hi @CodeOn-ArK 
The ordering of the payloads/blocks is dependent on the client library. You can read more about the OTA download protocol [here](https://docs.aws.amazon.com/iot/latest/developerguide/mqtt-based-file-delivery-in-devices.html#mqtt-based-file-delivery-get-getstream). The client can request `n` blocks starting from an offset or a bitmap of blocks. Server can return number of blocks up to the requested number along with the offset of each block. OTA client assembles the blocks based on offset.

ravibhagavandas, 2022-06-14:
Hope this answers your question.
I am closing this. For any further questions please use our [forums](https://forums.freertos.org/) for discussion.

closed 2022-06-14 by ravibhagavandas