aws/ota-for-aws-iot-embedded-sdk #456: [Feature Request] Misleading default config for otaconfigMAX_THINGNAME_LEN

dachalco, 2022-09-30:
**Is your feature request related to a problem? Please describe.**
AWS thingNames can be up to 128 length as documented [here](https://docs.aws.amazon.com/iot/latest/apireference/API_CreateThing.html). Yet, [by default](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/main/source/include/ota_config_defaults.h#L138) the library makes itself incompatible with all things of `nameLength > 64`.
```c
#ifndef otaconfigMAX_THINGNAME_LEN
    #define otaconfigMAX_THINGNAME_LEN    64U
#endif
```

**Describe the solution you'd like**
Correspondence with the AWS service.

**Describe alternatives you've considered**
User can override in their `ota_config.h`, but this is an _AWS provided_ library so it should alleviate user the need to research other AWS docs, to correctly override this library's default config.

**Additional context**
n/a

kstribrnAmzn, 2022-12-20:
First crack at this fix - https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/468

kstribrnAmzn, 2023-02-23:
Closing this out as this was fixed in https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/468. The new default max length is 128 characters. The only place this behavior is slightly different is when communicating the client token as the thingname is truncated to 53 characters. This is required (even without the 128 char length update) to avoid a ClientToken value which is [too long](https://docs.aws.amazon.com/iot/latest/apireference/API_iotfleethub_CreateApplication.html)

closed 2023-02-23 by kstribrnAmzn