aws/ota-for-aws-iot-embedded-sdk #457: [BUG] Submodule tinycbor not synced with v0.5.4 tag.

DavidJurajdaNXP, 2022-10-18:
**Describe the bug**
tinycbor commit:
https://github.com/intel/tinycbor/tree/9924cfed3b95ad6de299ae675064430fdb886216

referenced in 
https://github.com/aws/ota-for-aws-iot-embedded-sdk/tree/v3.4.0/source/dependency/3rdparty

is not pointing to tinycbor v0.5.4 tag as described in:

https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/v3.4.0/sbom.spdx
PackageName: tinycbor
SPDXID: SPDXRef-Package-tinycbor
PackageVersion: v0.5.4
PackageDownloadLocation: https://github.com/intel/tinycbor.git
PackageLicenseConcluded: MIT
FilesAnalyzed: True
PackageVerificationCode: da39a3ee5e6b4b0d3255bfef95601890afd80709
PackageCopyrightText: NOASSERTION
PackageSummary: NOASSERTION
PackageDescription: NOASSERTION

v0.5.4 tag is 
https://github.com/intel/tinycbor/commit/11590e470dfc1a144eb5665655c7ff6627b8e0f6

**Expected behavior**
Submodule synced with official tinycbor tag (v0.5.4).

**Screenshots**
[image]

archigup, 2022-10-18:
Thanks for letting us know! Looking into this.

archigup, 2022-10-18:
I've made a PR to fix this on the main branch

PR #458 (2022-10-18): Fix tinyCBOR submodule to match manifest
Fixes #457 on main.

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit 08513838 (2022-10-18): Fix tinyCBOR submodule to match manifest (#458)

Fixes #457 on main.
closed 2022-10-18 by archigup