aws/ota-for-aws-iot-embedded-sdk #460: [BUG] Issue with longer thing names

LozGMK, 2022-11-07:
Apart from the thing name length issue described in this other issue: [[Feature Request] Misleading default config for otaconfigMAX_THINGNAME_LEN](https://github.com/aws/ota-for-aws-iot-embedded-sdk/issues/456), there is another limit to thing name length in this code.

In `ota-for-aws-iot-embedded-sdk/source/ota_mqtt.c`, the thing name is added into string `pPayloadParts[ 3 ]`, which is part of a message published to AWS topic `$aws/things/[thing_name]/jobs/notify-next` that ends up something like `{"clientToken":"[decimal_value_of_reqCounter_in_a_string]:[thing_name]"}`. Since the decimal value of `reqCounter` can be up to 10 characters, the  thing name can only be 64 - 10 -1 = 53 chars before there is (potentially) a problem with this request, as AWS complains (and does not send back the requested data) if a client token longer than 64 characters is used.

If the intention is to place a unique value in `clientToken` (to tie the request and response together), then surely the value of `reqCounter` is sufficient, without also including the thing name?

As a side note, the value of `reqCounter` is turned into a decimal string using function `stringBuilderUInt32Decimal()`, which returns an empty string when it is sent a value of `0`, rather than `"0"`, as might be expected.

That function may cause issues for other uses of the function, such as with `versionMajorString`, `versionMinorString` and `versionBuildString`.

Tested on an ESP32.

AniruddhaKanhere, 2022-11-10:
Hello @LozGMK,

Thank you for bringing this to our attention. I'll inform the development team about this and will get back to you soon.

Thank you for your patience,
Aniruddha

AniruddhaKanhere, 2022-11-10:
@dachalco and @LozGMK,

I have created a PR to fix the string builder issue: https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/461 can you please take a look?

LozGMK, 2022-11-16:
I've had a look, put the same fix into my own code, and it seems to work fine.

LozGMK, 2022-11-24:
Were there any thoughts on the other issues?

Just to clarify, those are:
- that there is a misleading (and excessively limiting) default config value for `otaconfigMAX_THINGNAME_LEN`, limiting thing name length (value should be `128` - I think this is specified in two places in the code, though one may be `#define`d out)
- that adding the thing name to the client token sent to AWS (added into string `pPayloadParts[ 3 ]`, and published to AWS topic `$aws/things/[thing_name]/jobs/notify-next`, in file `ota_mqtt.c`), in the OTA demo, further limits the thing name length (no real need to add the thing name - use some other short string instead)

kstribrnAmzn, 2022-12-19:
Regarding the `thingname` length restriction, I agree with you and [the bug](https://github.com/aws/ota-for-aws-iot-embedded-sdk/issues/456) filed by @dachalco. This should match the maximum length defined by [AWS IoT Core](https://docs.aws.amazon.com/iot/latest/apireference/API_CreateThing.html).

As for adding the string name to the client token, I think a happy middle ground might be truncating the ThingName down to the first 53 (or fewer) characters so that the client token length restriction is never longer than the allowed 64 characters.

kstribrnAmzn, 2022-12-20:
I'm putting together a pull request (#468) to take a crack at fixing this. I have a couple more fixes to do - notably because I haven't worked with CMBC proofs before - and likely tweak my coding style a bit. I'll drop an update here when this is merged.

kstribrnAmzn, 2023-02-20:
PR https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/468 to correct the thingname length is accepted and merged. PR https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/475, which fixes the integer to string conversion is submitted. I'm going to close this out as I expect this PR to be approved shortly.

closed 2023-02-20 by kstribrnAmzn