aws/ota-for-aws-iot-embedded-sdk #464: [BUG] Issues when trying to mock these functions using googlemock.

phelter, 2022-12-06:
**Describe the bug**
When trying to build mocks for the SDK's API using GoogleTest - i.e. integration into C++ framework, there are several issues:

1. Headers do not have Extern C defined to ensure the names aren't mangled.
```cpp
#if defined(__cplusplus)
extern "C" {
#endif // defined(__cplusplus)
...
#if defined(__cplusplus)
}
#endif // defined(__cplusplus)
```
2. Use of `delete` in `ota_os_interface.h` - delete is a C++ reserved token - please rename.

**Host**
- Host OS: Linux
- Version: Ubuntu 18.04

**To Reproduce**
- Include any of the headers into a C++ file.
- Try and compiled file.  Watch the errors fly.

**Expected behavior**
- ability to use the API in a C++ environment as well. This is primarily for test.

**Screenshots**
```txt
 included from ../esp_aws_iot-src/libraries/ota-for-aws-iot-embedded-sdk/ota-for-aws-iot-embedded-sdk/source/include/ota.h:38:
../esp_aws_iot-src/libraries/ota-for-aws-iot-embedded-sdk/ota-for-aws-iot-embedded-sdk/source/include/ota_os_interface.h:272:22: error: expected member name or ';' after declaration specifiers
    OtaDeleteTimer_t delete; /*!< @brief Delete timer. */
```

**Wireshark logs**
N/A

**Additional context**
None

jasonpcarroll, 2022-12-06:
Hi @phelter,

Which files specifically don't have the C++ wrap? I took a look at ota_os_interface.h and it seems the delete is also covered at least on mainline https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/c3bd5840979cadfe1f9505e13e49cccb87333650/source/include/ota_os_interface.h#L281.

Best,
Jason Carroll

phelter, 2022-12-15:
Thanks @jasonpcarroll - I am now able to mock these using googlemock with the latest from `main` branch - the issue regarding missing C++ wrap is now exclusively an issue with [esp-aws-iot](https://github.com/espressif/esp-aws-iot) implementation of the port.  I've fixed that here - fyi: https://github.com/Retlek-Systems-Inc/esp-aws-iot/compare/master...bugfix/cpp-usage-and-other-fixes

This can be closed.

closed 2022-12-16 by phelter