aws/ota-for-aws-iot-embedded-sdk #486: [Feature Request] Use hardware accelerators for base64 operations

srnae05, 2023-04-06:
The OTA agent uses base64 decoding when downloading a file. This function is a pure software implementation. More often, microcontrollers have hardware accelerators that offer this functionality.

**Possible Solution**
The structure OtaInterfaces_t can be extended to set the interface for base64 operations. To support microcontrollers that do not have this possibility, the structure member could be set to NULL.
In function where the base64Decode() is called, the structure member is checked. If it is not NULL, use hardware accelerator API, else use the software implementation in the OTA SDK

**Alternatives**
The base64_decode function in the SDK could be configured as a WEAK function, offering the possibility to the application to implement a strong function (using hardware accelerators).

johnrhen, 2023-04-06:
Thanks for the feature request - we'll put it in our input queue and prioritize accordingly.

n9wxu, 2023-12-20:
No further development will be performed on this library. See the readme.

closed 2023-12-20 by n9wxu