aws/ota-for-aws-iot-embedded-sdk #490: [BUG]

arkhipenko, 2023-06-02:
**Describe the bug**
An MQTT OTA request made by the library to AWS usinf CBOR encoding results in the Invalid response on the /rejected topic

**Host**
- Host OS: esp32
- Version: latest library code as is on github

**To Reproduce**
- Initiate OTA update 
- Run on esp32 and observe all CBOR requests failing

**Expected behavior**
I was expecting to start receiving OTA file chunks on MQTT topics.

**Additional context**
Initial request is made successfully
```
00:00:00:05.252 T: prvMqttPublish()
{"clientToken":":840d8ee419b8"}
7b 22 63 6c 69 65 6e 74 54 6f 6b 65 6e 22 3a 22 3a 38 34 30 64 38 65 65 34 31 39 62 38 22 7d 
```
AWS IOT responds with the job document as expected and a few ota methods are invoked as expected:
```
00:00:00:05.345 V: subscribeCB: Data
	topic: $aws/things/840d8ee419b8/jobs/$next/get/accepted
	payload: {"clientToken":":840d8ee419b8","timestamp":1685747465,"execution":{"jobId":"AFR_OTA-OTA_TEST-230602-1828","status":"QUEUED","queuedAt":1685747355,"lastUpdatedAt":1685747355,"versionNumber":1,"executionNumber":1,"jobDocument":{"afr_ota":{"protocols":["MQTT"],"streamname":"AFR_OTA-93e1bb19-0dcd-4aeb-9946-5df0260dc063","files":[{"filepath":"/ota/file","filesize":127762,"fileid":0,"certfile":"/pem","fileType":0,"sig-sha1-rsa":"na"}]}}}}
00:00:00:05.397 V: subscribeCB: handling JOB_ACCEPTED_RESPONSE_TOPIC
00:00:00:05.407 T: otaPal_GetPlatformImageState()
00:00:00:05.409 T: otaPal_CreateFileForRx()
00:00:01:26.993 T: otaPal_GetPlatformImageState()
00:00:01:27.000 V: prvMqttSubscribe: subscribed to topic $aws/things/840d8ee419b8/streams/AFR_OTA-93e1bb19-0dcd-4aeb-9946-5df0260dc063/data/cbor
```

then a request for the first chunk is published:
```
00:00:01:27.005 T: prvMqttPublish()
¦accrdyaf al ao abDÿÿÿÿan
a6 61 63 63 72 64 79 61 66 00 61 6c 19 10 00 61 6f 00 61 62 44 ff ff ff ff 61 6e 01 
00:00:01:27.028 V: publishMessage: topic   : $aws/things/840d8ee419b8/streams/AFR_OTA-93e1bb19-0dcd-4aeb-9946-5df0260dc063/get/cbor
```
this message decoded is:
`{"c": "rdy", "f": 0, "l": 4096, "o": 0, "b": h'FFFFFFFF', "n": 1}`
according to this: https://cbor.me/

the response is:
```
00:00:01:27.149 V: subscribeCB: Message arrived
00:00:01:27.150 V: subscribeCB: Data
	topic: $aws/things/840d8ee419b8/streams/AFR_OTA-93e1bb19-0dcd-4aeb-9946-5df0260dc063/rejected/cbor
	payload: ¿dcodekInvalidCboraokInvalidCborgmessagetInvalid CBOR messageamtInvalid CBOR messageÿ
```

This is consistent with the failure I reported for the MQTT download agent library here:
https://github.com/aws-samples/aws-iot-mqtt-download-agent/issues/3

Can someone please look into this? The CBOR message seems to produce the bitmap that is rejected by the backend!

gmtt, 2023-06-04:
Thank you for bringing this to our attention! We will be looking further into this issue as soon as we can, thank you for your patience

arkhipenko, 2023-06-06:
@gmtt - thank you - do you have an estimate of when we can expect some answer/resolution? We are in the middle of implementation and were hoping that the AWS library will work with the AWS backend "as-is". This is quite unexpected honestly.
We need to make a decision on whether to continue using the AWS OTA library or move to some other solution (which would be very unfortunate to be honest).

gmtt, 2023-06-07:
Hi @arkhipenko, the encoded hex appears to be correct. However, we suspect that the issue may lie in the cloud part. Could you please enable debug logging to obtain more information about this? Additionally, could you try using the tinycbor version that we tested here: https://github.com/aws/ota-for-aws-iot-embedded-sdk/tree/main/source/dependency/3rdparty instead of the latest version? As we haven't tested the latest version of this library, it's possible that an incompatible library is causing the problem.

arkhipenko, 2023-06-07:
Hi @gmtt 
Thank you for getting back to me. 
Based on the observed behavior, I do also suspect that the problem is with the cloud end of the process. 

Meanwhile: 
I have downgraded CBOR library to the version you advised me to work with (0.5.4) 
I have enabled all logging and attached is the interaction log from the last session (a few minutes ago) with the same unfortunate outcome. 
[ota-poc.log](https://github.com/aws/ota-for-aws-iot-embedded-sdk/files/11678592/ota-poc.log)

arkhipenko, 2023-06-09:
Hi @gmtt 
Just checking about the status of this - we are on hold until we hear back from you as to whether to expect a fix and when. 
Our assumption with choosing of this library was that it should "just work" b/c it is a library provided by AWS communicating with the AWS backend. It is quite surprising to discover that this is not the case.   I would much prefer it was a bug in our code since that is easy to fix.  Please advise!

gmtt, 2023-06-09:
Hi @arkhipenko
Sorry for the delayed response. We are currently investigating the issue and communicating with the cloud team to identify the problem. Please note that it may take some time due to our bandwidth limitations. We appreciate your patience.

gmtt, 2023-06-14:
Hi @arkhipenko
We have used our other tool to perform the same task and encountered the same error, indicating a strong likelihood that the problem lies in the cloud component. We have already informed our cloud team about the issue and requested an investigation. We will provide you with an update as soon as we have identified the cause.

arkhipenko, 2023-06-14:
Thank you @gmtt 
What is surprising to me is that there were no 1000s of reports of MQTT based OTA not working. I was expecting this to be used a lot!

AniruddhaKanhere, 2023-06-15:
Hello @arkhipenko,

Firstly, apologies that it is taking a bit longer than expected.

I tried with the older release version of OTA library and my firmware upgrade worked as expected.
Can you also try using the last released stable version of the library as here: https://github.com/aws/ota-for-aws-iot-embedded-sdk/releases/tag/v3.4.0?

We are working on figuring out what is causing the issue you are seeing.
This experiment would let us know whether there is anything wrong in the configuration that you are using to build/compile the library.

Thank you,
Aniruddha

arkhipenko, 2023-06-16:
I tried compiling with releases 2.0.0 and 3.0.0, but there are too many binary API mismatches - it would take a long time to adjust.

AniruddhaKanhere, 2023-06-16:
Hey @arkhipenko, I was trying the projects in one of our reference integration projects in this repository: https://github.com/FreeRTOS/iot-reference-nxp-rt1060

It uses the OTA lib submoduled here: https://github.com/FreeRTOS/iot-reference-nxp-rt1060/tree/main/Middleware/AWS. The version in use is v3.4.0.

One of my colleagues also mentioned that it worked for them when they tried it on Tuesday (13th June).

That being said, we are looking into the cloud side of things as well to make sure that there is no breaking change there.

Thank you for your patience.

arkhipenko, 2023-06-16:
@AniruddhaKanhere - could you please confirm that your colleague tested MQTT-based OTA, and not HTTPS-based (which works fine for me as well)?  I struggle to understand how the same code base produces different results - I do not modify the library code in any way!   

I would also very much like to see the CBOR message sent to the cloud from that test on June 13th, and the response from the cloud. Would your colleague have those by any chance?
Thank you!

arkhipenko, 2023-06-16:
@AniruddhaKanhere - just tried the exact version that you mentioned with the same result:

```
00:00:01:27.954 T: prvMqttPublish()
accrdyafalaoabDan
a6 61 63 63 72 64 79 61 66 00 61 6c 19 10 00 61 6f 00 61 62 44 ff ff ff ff 61 6e 01 
```

followed by the same 
```
00:00:01:28.056 V: subscribeCB: Message arrived
00:00:01:28.057 V: subscribeCB: Data
topic: $aws/things/840d8ee419b8/streams/AFR_OTA-2ee1fa35-479b-4210-8b1b-547f8765a3dd/rejected/cbor
payload: dcodekInvalidCboraokInvalidCborgmessagetInvalid CBOR messageamtInvalid CBOR message
bf 64 63 6f 64 65 6b 49 6e 76 61 6c 69 64 43 62 6f 72 61 6f 6b 49 6e 76 61 6c 69 64 43 62 6f 72 67 6d 65 73 73 61 67 65 74 49 6e 76 61 6c 69 64 20 43 42 4f 52 20 6d 65 73 73 61 67 65 61 6d 74 49 6e 76 61 6c 69 64 20 43 42 4f 52 20 6d 65 73 73 61 67 65 ff 
00:00:01:28.101 T: handleMessage()
```

arkhipenko, 2023-06-20:
@AniruddhaKanhere - it is quite interesting. 
I had another look at the payload and as far as I can understand the bitmapping, every bit of the bitmap represents a block of data.  What was your colleague's file size? 
According to my stream definition: `[{"filepath":"/ota/file","filesize":127762` the file size is 127762 bytes. 
That size represents 31.19 blocks 4096 bytes each., so a 32 bit bitmap (or 8 bytes) should be sufficient, which explains the number of 'F's in my 'b' field CBOR message:
```
   44          # bytes(4)
      FFFFFFFF # "\xFF\xFF\xFF\xFF"
```

a 20-byte 'b' field would suggest a file 655360 or so bytes long. 
Regardless, it is still perplexing why his message is accepted and mine is rejected.

AniruddhaKanhere, 2023-06-21:
> I had another look at the payload and as far as I can understand the bitmapping, every bit of the bitmap represents a block of data. What was your colleague's file size?

Yes, that is what the bitmap means. But I don't think that the size of the file should matter because we are still requesting the file and while doing that, we do not know the size of the new file. For all it matters, the new file can be double the size and may not fit in the bitmap that we sent at all. I am not sure though so don't quote me on this :)

But, the image is quite big as you predicted: `34 28770 [OTAAgent] [INFO] Extracted parameter: [key: value]=[filesize: 647324]`.

That being said, the bitmap has a `7F` at the end in his case.

By the way, we were able to figure out that there is nothing wrong with the cloud side of things. We were just sending wrong CBOR.

But this is good news because it now pinpoints that there might be something wrong on the application side of things. And we can figure out what is wrong in your application. :)

Thanks,
Aniruddha

arkhipenko, 2023-06-21:
When you say "we are sending the wrong CBOR", what does it mean exactly? 
What would the  _right_ CBOR be for that scenario? (Given that you have the exact file stream information). 
@AniruddhaKanhere

AniruddhaKanhere, 2023-06-21:
Ah yes, let me clarify.

If we were supposed to send `A6 F1` (assume it to be valid CBOR), then we were actually sending the binary for ASCII `A` (decimal 65) then binary for ASCII `6`(decimal 54) then binary for ASCII `F`(decimal 70) and then binary for ASCII `1` (decimal 49).
We were supposed to send 0xA6 (decimal 166) and `0xF1` (decimal 241).

This happened because we assumed that the test setup we were using was aware of the hexadecimal numbers. So when we sent `0xF1`, we assumed that it would send 241 in decimal notation. But, that test setup wasn't hexadecimal system aware and it actually sent `0..x..F...1`.

But, when we did send the valid CBOR message, it was accepted by the cloud.

Following up on your issue though; would it be possible for you to upload your code (only the parts we need to understand this issue) to github and share with us? That way we can replicate your results.

arkhipenko, 2023-06-22:
@AniruddhaKanhere 
Thank you for your message.
I fully appreciate how the CBOR messaging works. What surprises me is that I literally use CBOr library 0.5.4 as-is, no modifications, AWSIOT library code 3.4.0 as-is, and the results are rejected. I cannot explain this. 
The library does not produce '0x' prefixed bitmap, it produces an 'h' prefixed bitmap, which seems to be rejected by the backend. I am yet to see a reasonable explanation why that is. There is barely any code in want you refer to as 'my code' - it just facilitates the MQTT messaging via standard esp32 library. The rest is 100% unmodified AWS, Core JSON and CBOR code. If the 'h' prefixe before the bitmap is a problem, how do I make CBOR not prefix it ?

arkhipenko, 2023-06-22:
@AniruddhaKanhere 
Thank you for your message.
I fully appreciate how the CBOR messaging works. What surprises me is that I literally use CBOr library 0.5.4 as-is, no modifications, AWSIOT library code 3.4.0 as-is, and the results are rejected. I cannot explain this. 
The library does not produce '0x' prefixed bitmap, it produces an 'h' prefixed bitmap, which seems to be rejected by the backend. I am yet to see a reasonable explanation why that is. There is barely any code in want you refer to as 'my code' - it just facilitates the MQTT messaging via standard esp32 library. The rest is 100% unmodified AWS, Core JSON and CBOR code. If the 'h' prefixe before the bitmap is a problem, how do I make CBOR not prefix it ?

Can you actually pinpoint what is wrong with my CBOR message?:

```
A6             # map(6)
   61          # text(1)
      63       # "c"
   63          # text(3)
      726479   # "rdy"
   61          # text(1)
      66       # "f"
   00          # unsigned(0)
   61          # text(1)
      6C       # "l"
   19 1000     # unsigned(4096)
   61          # text(1)
      6F       # "o"
   00          # unsigned(0)
   61          # text(1)
      62       # "b"
   44          # bytes(4)
      FFFFFFFF # "\xFF\xFF\xFF\xFF"
   61          # text(1)
      6E       # "n"
   01          # unsigned(1)
```

arkhipenko, 2023-06-22:
@AniruddhaKanhere - would it be possible for me to add you to a private Git repo for code sharing?
Also, this is written for esp32 - do you have an esp32 dev board you can run it on?

AniruddhaKanhere, 2023-06-22:
Hey @arkhipenko,

I am not saying that anything is wrong with your CBOR as it is, maybe we are missing something.

Can you try the steps mentioned [here](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/GettingStartedGuide.md) in this repository: https://github.com/FreeRTOS/iot-reference-esp32c3?
This is one of our reference integration repositories and the OTA on that should just work if you follow the instructions in the getting started guide that I linked above. Can you please try that and let us know what happens?

I suggest using everything new for this test (including the 'thing' and corresponding certs/keys and policies). All the process should be outlined in the above linked GSG.

> would it be possible for me to add you to a private Git repo for code sharing?

I think it should be fine if you think there is nothing proprietary in there which I should not see.

arkhipenko, 2023-06-22:
@AniruddhaKanhere - after careful re-examination of my code I found a nasty little bug in the way outgoing MQTT messages were handled corrupting the CBOR message at the very last step of the process. Corrected and I seem to be getting the MQTT file payloads as expected now. 
I apologize for tormenting you with what indeed ended up being an issue on my end, albeit not trivial. 
Thank you for your help.
I will close this issue now and open a new one if there are other problems. 
Thank you again!

closed 2023-06-22 by arkhipenko

AniruddhaKanhere, 2023-06-22:
It is great news that you found the bug @arkhipenko!
Thank you for informing us about it. Would you mind elaborating what was happening in your MQTT sending logic? I ask because if this is common enough, then it would benefit all readers of this issue :)

Thanks again for informing us about the issue and the fix,
Aniruddha

arkhipenko, 2023-06-22:
The original MQTT code assumed payload was always an ASCII string, so the very final buffer copy was done with `strncpy()` method, which of course is not suitable for true binary payload, as it stops at the first zero. 
Replacing that with `memcpy()` solved the problem. 
So everything looked correct and was correct right up to the point of sending to the backend. 
It's embarrassing to admit, but those are the bugs you don't think about... 
Again, thanks for your help!