aws/ota-for-aws-iot-embedded-sdk #491: Read outside of Json buffer

gonzzor, 2023-06-03:
The printf format to LogDebug expects pJson to be null-terminated, but its length is given by messageLength. Thus it will read more than expected.

https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/406415ef38d9f0fbabd69735623008fd090043b3/source/ota.c#L1821

Something like this should fix the issue.
`    LogDebug(("JSON received: %.*s", messageLength, pJson));`

gmtt, 2023-06-04:
Thank you for bringing this to our attention! We will be looking further into this issue as soon as we can, thank you for your patience

ActoryOu, 2023-07-27:
Hi @gonzzor,
Sorry for late response. The PR is created to apply this change back to OTA library.

kstribrnAmzn, 2023-07-31:
The PR - https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/493

ActoryOu, 2023-08-09:
I'm closing this issue since PR's merged

closed 2023-08-09 by ActoryOu