aws/ota-for-aws-iot-embedded-sdk #509: [DOC] What is the current state of this library?

TorstenRobitzki, 2024-03-26:
Following the link from the `aws-iot-device-sdk-embedded-C` repository, I got the information here, that the library will be deprecated in 2024, which basically signals to me, that basing a new project on this library is probably not a good idea.

- Why is there is still the link to this library from the `aws-iot-device-sdk-embedded-C` library, if this library will be deprecated soon?
- It looks like there is active development at this library, but there was no new version released in the last 2 years?

Assumed, I want to have a take on the "AWS IoT Over-the-air Update" feature, how would I start a new development?

tony-josi-aws, 2024-03-27:
@TorstenRobitzki 

Did you take a look at the [FreeRTOS OTA page](https://freertos.org/freertos-core/over-the-air-updates/index.html)? There you can find the OTA [reference demo](https://freertos.org/freertos-core/over-the-air-updates/mqtt-ota-agent-orchestrator.html) using OTA Agent Orchestrator: [Lab-Project-ota-example-for-AWS-IoT-Core](https://github.com/FreeRTOS/Lab-Project-ota-example-for-AWS-IoT-Core)

> Why is there is still the link to this library from the aws-iot-device-sdk-embedded-C library, if this library will be deprecated soon?

As the library is not yet deprecated and there are demos/projects still referencing the older library, the OTA library is still linked in the `aws-iot-device-sdk-embedded-C` repository.

> It looks like there is active development at this library, but there was no new version released in the last 2 years?

The changes after the last release are essentially bug fixes, and there aren't any major feature additions as such.

TorstenRobitzki, 2024-03-27:
@tony-josi-aws Thank you for your response!

closed 2024-03-27 by TorstenRobitzki